WandererFan/FreeCAD-TechDraw #1: Add Python Scripting

WandererFan, 2016-01-18:
Python scripting support is incomplete.  Need to create/update *Py.xml and *PyImp.cpp files

WandererFan, 2016-01-30:
Python functionality implemented for:
Page
DrawViewPart
DrawProjGroup
DrawViewSection
DrawViewAnnotation
DrawViewDimension
DrawViewSymbol

SaveAsSvg is not yet implemented as it is part of the MDI View, not the FC Document.  (ie I don't know how to do it yet.)

WandererFan, 2016-01-30:
oops.

WandererFan, 2016-01-30:
Test scripts are in [gihub](https://github.com/WandererFan/FCDraw2Test/Scripts)

ianrrees, 2016-04-19:
Have added a saveSvg() Python method on the page object, but there's still some old code in AppTechDrawGuiPy.* that we should probably clean out.  There's some SVG import functionality there, I'm assuming that will now be covered using the "Symbol" feature?

Do you know if we have/want capability to export DXF?

WandererFan, 2016-04-19:
AppTechDrawGuiPy is just what was in Drawing with enough changes to get a
compile, isn't it?

The Drawing SVG import brought in a whole drawing.  I haven't tried that
with Symbol, but I don't see why it wouldn't work (you'd have to use a
blank template as a starting point).

What did people do with the old DXF export?  Transfer drawing to AutoCAD?

ianrrees, 2016-04-19:
Yes, I think AppTechDrawGuiPy is a leftover from the original Drawing.

Not sure what people would want DXF export for - my guess would be machines like pen plotters or laser cutters might want it?  I'll try to figure out how hard it would be to support, and remove AppTechDrawGuiPy anyways, as I don't think it will be useful in that form.

WandererFan, 2016-04-20:
OK. Can always add it back if we come up with some convenience functions
for scripting.