WandererFan/FreeCAD-TechDraw #18: Finish edition of template texts

ianrrees, 2016-01-30:
Moving this over from ianrrees/FreeCAD_tinkering - was made ~6 months ago, so there's a good chance some of these items have been taken care of.

  * Python interface - changing strings, also setting view orientation while there
  * Show PropertyMap in PropertyEditor?
  * Escape user strings + handle SVG errors better (currently throws unhandled exception)
  * DXF support?
  * Add blank template option, would be good for error cases like above
  * Bigger minimum size for template string dialog
  * Don't draw dotted box on SVG render

WandererFan, 2016-01-31:
- dotted box is now a green edit handle.
-A?_Landscape_plain.svg for blank template??

WandererFan, 2016-05-16:
"Python interface - changing strings, also setting view orientation while there"
  - Py functions for getting/setting EditableText strings added. 
  - Don't understand view orientation for Template
"Bigger minimum size for template string dialog"
  - Real QDialog with room for fieldnames/content added.  QInputDialog::getText is too hard to beautify.
"Show PropertyMap in PropertyEditor?"
  - don't need for TechDraw.  eventually all property types should be in editor
"Escape user strings + handle SVG errors better (currently throws unhandled exception)"
  - don't understand this one.

ianrrees, 2016-05-18:
View Orientation - I'm not sure if it's important anymore, but there's a Property associated with the template orientation.  Was thinking we should have Python methods for manipulating that (or do those come for free?)

Escaping user strings and handling SVG errors - For instance, try changing a drawing/template title to "Hello

WandererFan, 2016-05-18:
App.ActiveDocument.myPage.Template.ViewOrientation = "Landscape" should
work (or ... = 1 since it is an enum).

Might need recompute() after to see results.

OK, I understand the SVG thing now.

Thanks.

On Wed, May 18, 2016 at 4:16 AM, Ian <notifications@github.com> wrote:

> View Orientation - I'm not sure if it's important anymore, but there's a
> Property associated with the template orientation. Was thinking we should
> have Python methods for manipulating that (or do [...]

WandererFan, 2016-05-20:
"DXF support?" 

- I'm going to say leave this for now.  There are DXF to SVG converters if somebody really needs to use a DXF based template.

WandererFan, 2016-05-24:
"Escape user strings + handle SVG errors better (currently throws unhandled exception)" 

*** fixed

WandererFan, 2016-05-24:
Everything on this list except:
- PropertyEditor changes fore PropertyMap
- DXF template support 
is fixed, so I'm going to close this issue.

closed 2016-05-24 by WandererFan