WandererFan/FreeCAD-TechDraw #19: Rendering of large-scale objects

ianrrees, 2016-01-30:
Things get really slow when a Drawing of a really big sphere (r=500km) is requested - investigate this.

From vague memory, the issue is that we initially try to draw the item at document scale, and the canvas gets expanded to some absurd pixel dimension (which takes the time).

WandererFan, 2016-07-17:
Autoscale at creation?  if BBox >> PageSize,  Scale == 1/(BBox/PageSize)

What about if BBox << PageSize?  Magnify it?

ianrrees, 2016-07-17:
I like the idea of defaulting to auto scale when adding items to the drawing in general.  Maybe it using the logic from the current auto scale that picks "standard" scaling values.

Agree that small items should scale up.  IIRC that is less of a problem since it doesn't cause the slowdown.  Do we run in to problems with tolerances on selection of really small items?

WandererFan, 2016-07-18:
Hey, good to here from you! I was just thinking about sending a ping.

Scaling to best fit standard size sounds good.

Should be ok with the OCC stuff.  It mostly uses Precision::Confusion(), I
think.  Was never 100% sure what the TD Precision property was for.

WandererFan, 2016-08-18:
fixed.  master commit cf90d6931914bb1f7a0cdb1eefd27b14e23bf458

closed 2016-08-18 by WandererFan