WandererFan/FreeCAD-TechDraw #23: Ability to Produce Drawings Without /Gui

WandererFan, 2016-02-03:
TechDraw/App will not produce a drawing on its own.  TechDraw/Gui is required to provide the QGraphicsScene which contains the actual drawing.  This seems in conflict with FreeCAD's principle of being able to work in a non-Gui mode.

ianrrees, 2016-02-12:
Very true - I'll look at separating the GUI bits from the QGraphicsScene.

ianrrees, 2016-02-12:
Acceptance tests for this:
  1. Able to build TechDraw with BUILD_GUI=0 which doesn't include any files in the Gui directory.  This currently works for TechDraw, but not for a few other modules in FreeCAD...
  2. Able to output an SVG (pdf?) from TechDraw of a Part Design object via a Python script run in FreeCADCmd built with BUILD_GUI=0.
  3. 1 and 2 imply that non-GUI code that's currently in src/Mod/TechDraw/Gui are moved to App directory.

ianrrees, 2016-02-14:
I moved one class following the pattern in my email as a test case.  The diff is pretty confusing...

https://github.com/ianrrees/FreeCAD_tinkering/commit/e1791c05f3599e6c2e9b0894fc5099b6b21622c7?diff=split

WandererFan, 2016-02-14:
> ianrrees/FreeCAD_tinkering@e1791c0?diff=split

At some point in a change this big, we're going to have to just delete all
the old stuff and replace everything.   Like we did with luke's stuff and
the Drawing->TechDraw conversion.

ianrrees, 2016-04-19:
Got to the point where we can make drawings via Python with BUILD_GUI=0 and export them to SVG.  It turned out to be a bit of a pain to spin up a QGraphicsScene without the GUI going - there's an AFACT undocumented dependency on Qt's event loop, which isn't running if FreeCAD was built without GUI.  See dd03d4a70f43a695fe892e1287e33126755ea564 for the workaround.

Currently, can only include templates in the drawings, but I think it shouldn't be too hard to move the other QGraphicsItems code over.

WandererFan, 2016-04-19:
Fuzzy area for me, but isn't the event loop part of QtCore/QApplication?

ianrrees, 2016-04-19:
Yes, and I think it might be worth discussing with the main developers whether we want the FreeCAD command line application to have a Qt event loop running.  Certainly would make some things easier, and I can't think of any major problems with doing that.

WandererFan, 2016-04-20:
I'll leave that to you.  I'm no expert in Qt arch.

WandererFan, 2016-05-01:
Note that Clip is currently broken due to this issue.  Y coordinates not updated. See GIBase::getYInClip().

WandererFan, 2016-07-12:
This issue appears to be insurmountable at this point due to TechDraw's tight coupling with QGraphics and QtSvg modules.

On indefinite hold.

closed 2016-07-12 by WandererFan