WandererFan/FreeCAD-TechDraw #27: How to handle "Document" scaling (and projection type, etc)

ianrrees, 2016-02-12:
In the current Projection Group (and perhaps other places?), there's an option for scale "Document", which I find confusing and am not sure what exactly we want it to do.  It seems like the plan was to be able to drive scaling from the document level, but not sure how useful that would be.  Should we just remove it?

WandererFan, 2016-02-12:
I've always understood it to be a default for the all the Views on the
Page.  Custom is for a single view in a different scale - like a detail.

"Page" is probably a better name than "Document"

ianrrees, 2016-02-12:
Agree that "Page" is a better name.

I imagine that setting scaling to auto (at either page or object level) as a default would be best, rather than the current 1:1?  So, presumably we'll need some way for the auto scale to work at a page level to allow for a smart scaling when there are multiple page-scaled objects in the same auto-scaled page.

We'll also need to consider linking the scale property for the page with the scale text in the title block.

WandererFan, 2016-02-13:
Honestly, never looked into "Automatic".   Is it supposed to make a "best
fit" on the page?

ianrrees, 2016-02-13:
I re-worked the automatic scaling for the projection groups - IIRC it essentially tries to fit the projection group into the sheet with the largest possible "nice" scaling value.

If we were going to try scaling to fit multiple objects at the same time (which I think would be quite nice from a UI standpoint!) then we'd need a more sophisticated method that would work at page level.