WandererFan/FreeCAD-TechDraw #28: Detail, cross section, broken out section, hole table

efyx83, 2016-02-12:
Hi,
here are some usefull features and would be great to see them in your module:
- detail,
- cross section,
- broken out section
- hole table (for example: https://www.solidsolutions.co.uk/Uploaded/Image/BLOG-PICS/BarryChatham/solidworks-HoleTable-Blog.png)

I know that you have worked on cross section, but I left it on list. Anyway hopefully you will like requested features. Will you able to provide those points?

ianrrees, 2016-02-12:
Goal at this point is to get a solid base working, and we'll see where things go from there.

In the meantime, it would be great to get feedback from testing what we've done already.

WandererFan, 2016-02-12:
The hole table should be relatively straight forward once we get SpreadsheetView implemented. #24 
Detail view might be a Clip with a larger scale value? 
Not sure what a "broken out section" is.  Link?

Thanks for the input.

WandererFan, 2016-02-12:
[found a "broken out section"](https://www.me.utexas.edu/~rbarr/draft/Section_Views/Broken-Out_Section.html)

ianrrees, 2016-02-14:
@efyx83 - could you please elaborate on what you mean by "detail"?  If you don't mind, I'd like to split this issue into 4 separate issues.

efyx83, 2016-02-14:
@ianrrees sure. Drawing detail is very often using and needed "feature". It is for zoom in some part of object. You are selecting region by circle (or different shape) after that you have zoomed additional view. Selected circle is marked by letter (next free) A, B, C etc.. and zoomed view is under described with: Detail A, B, C... and a scale of zoom. Hopefuly this is written quite clearly, in case if not ;) see attached link: http://blogs.solidworks.com/solidworksblog/wp-content/uploads/sites/2/2014/01/1-30-2014-12-53-57-PM.jpg
Video: https://www.youtube.com/watch?v=92XDSTsFr-0 (around 0:45s)

"Broken out section", you can draw a spline into region and then define depth, then you are getting additional view with "broken". I'll add video of process in SW
https://www.youtube.com/watch?v=2qj2J4QbWx8 (about 2:10min is a resault)

ianrrees, 2016-02-15:
I've created 4 separate issues to track the parts of this one.

closed 2016-02-15 by ianrrees