WandererFan/FreeCAD-TechDraw #30: Precompiled headers

ianrrees, 2016-02-14:
I suspect that the PreCompiled.h in TechDraw has become a bit of a mess, but not sure what a good way to check is.

Maybe bang together a little script to find everything between the #ifndef _PreComp_ blocks and use it as the base of a cleaned up PreCompiled.h ?

WandererFan, 2016-02-14:
Blush, guilty as charged.

The include lists in general seem to have everything but the kitchen sink
in places.

I'd have to write a Python program or something for this.  If you're adept
with grep & co that might be faster?

ianrrees, 2016-02-15:
No worries, I think it's fine to keep chugging away with coding, then go back and clean up the #includes later.  IIRC, the only platform that uses the precompiled headers is Windows, so who knows what that'll look like ;).

I'd personally tackle this with Python too, don't think it would be too much trouble.  With Qt's nice naming of headers and classes being the same, I bet we could cleverly detect instances of Qt headers that are included without using the class they provide.

WandererFan, 2016-05-29:
I have several instances where an OCC file is included in PreCompiled.h, but I have to add it in my cpp file to pick it up.  I don't know why.

Also I had to add OCC Standard_Version.hxx to TechDraw/App/PreCompiled.hxx

WandererFan, 2016-07-14:
fixed FreeCAD/master de76cd6b53e5d3f4b9bdf662e1811fa8aff1d9c7

closed 2016-07-14 by WandererFan