WandererFan/FreeCAD-TechDraw #33: Provide for "Cross Section" Views

ianrrees, 2016-02-15:
Splitting this one out from https://github.com/WandererFan/FreeCAD-TechDraw/issues/28

WandererFan, 2016-02-15:
Is this different from DrawViewSection?  or is it an enhancement to DrawViewSection?

ianrrees, 2016-02-15:
Not sure - TBH I haven't actually used the section feature yet.  Just wanted to split up the original bug into more manageable pieces to deal with them individually.  Perhaps this one is done already?

@efyx83 - have you tried the section view feature, and if so is there anything that should change about it?  Thanks!

efyx83, 2016-02-15:
I'll try today somehow install your module into FreeCAD. I belive I need compile whole thing into one - is that better on linux? I will be happy to make some test with your module :)

ianrrees, 2016-02-15:
It'll be easiest to build all of FreeCAD including this module, rather than just building the module.  It should be possible to build on Windows/Linux/MacOS.  WandererFan uses Linux and I use MacOS, so if you were to try Windows you might encounter some problems we haven't seen yet.

Instructions for either platform are on the FreeCAD wiki - you'll essentially want to follow the normal instructions, except instead of cloning from FreeCAD/FreeCAD, clone from WandererFan/FreeCAD-TechDraw and checkout the TechDraw branch before building.

efyx83, 2016-02-23:
Hi, finally I find some time but I was defeated by compiling of your version of FreeCAD.
1st I have downloaded files and unpack into folder: https://github.com/WandererFan/FreeCAD-TechDraw/tree/TechDraw
2nd I tryed to cmake, got answer "Python not found"
3rd I checked Wiki: http://www.freecadweb.org/wiki/index.php?title=CompileOnUnix#Compile_FreeCAD and there is to install (I have linux mint so ubuntu repository) a lot of packages. But some of them are installing some not. I have Python already installed but still cmake shows "Python not found"

Is there any easier way to make FreeCAD from your code? Version for dummies ;)

PS. Sorry for making mess in that place, but I have no idea where to ask.

WandererFan, 2016-02-24:
Can you compile 'master' on Mint?  There shouldn't be any extra
complications for TechDraw, but let's eliminate the variables 1 by 1.

efyx83, 2016-02-25:
I'll check today's evening. I am not good in that, actually I don't understand vell how the git works but..
I need to clone 'master' from https://github.com/FreeCAD/FreeCAD.git free-cad-code and cmake it. This should compile without any problems. I understand your module is included and will be available after compilation. Am I right?

ianrrees, 2016-02-25:
If you clone the master from that URL (the main FreeCAD one), then you will _not_ have the TechDraw module.  But, I do think that is a good test - if you are able to build the main FreeCAD, then you should also be able to build the version of FreeCAD in this repository (this has TechDraw).

I've added a small section to the wiki [here](https://github.com/WandererFan/FreeCAD-TechDraw/wiki) which might help clarify.

WandererFan, 2016-03-24:
fixed. DrawViewSection

closed 2016-03-24 by WandererFan