WandererFan/FreeCAD-TechDraw #38: First impressions

efyx83, 2016-03-09:
Hi,
after many hours at least I've got compiled freecad with TechDraw in it ant these are my first impressions.
1) What I liked is projection of model - that you can turn it, rotate and then add rest views. This is really cool - good control. Not necessary it is comfortable, first to click on model, then to "projection" button and then go to drawing tab to do the projection (btw. you cannot move it around until you click "OK"). It would be better to click first "projection button" then it will ask which model add to drawing.
2) Good is that you can choose scale while "projection". Not necessary good is that naming under view for example: "front", "bottom" etc.. good I can remove that. Instead that, it will be good when you are choosing different scale than "page scale", should text appear something like this for example "Scale 1:1", because if scale is set for page you can add to drawing other views with different scale and then under text is necessary.
3) I don't like that I need to choose horizontal or vertical dimensioning. Why? It should automatically decide is it vertical or horizontal or between corners.
4) Next thing when you are rotating "main" view by custom angle (that's great) you should have option to project object by this angle - and here we are back again in projection which is not friendly. You should choose view then projection button and freely add new view by pulling it out.
5) You can move views, that's a large step but if you click on left side menu to "X", "Y", "tolerance" it is going back on place (destroys whole view sets).
6) I don't like that you can go out with a view over the edge of "paper" but you cannot pan. Looks like card is centred on a screen and cannot move over grey area.
7) Unfortunately I cannot use cross section. Can you describe how to use that?
8) I don't know how to use "hatched area"
9) Hidden lines should be dashed not light grey. I mean can be light grey but needs to be dashed ;)

PS. Hopefully I am understandable and you liked my comments. O/A good job ;) . Please answer for point 7 and 8, that will help me to test those features. 
When do you think will be new features (as I wrote before?), how I'll be able to update my freecad TechDraw next time (again cloning and compiling whole?)
PS2. Weld symbols are also necessary it is quite good done on Hamish "Drawing dimensioning", also he have some intelligent dimensioning - maybe you can see some of his code.
Cheers.

efyx83, 2016-03-10:
Next feature is thread mark. It is little hard because it would be perfect to connect this with "hole creator" - there isn't any I believe right now. It could be perfect when creating threaded hole, you would choose thread size (for example ISO M6) and get hole under M6 thread and on drawing automatically it would be shown as 3/4 of circle (thin line) and it is turned a little (not sticking center cross (BTW center line and center cross is needed also ;) )) See attached drawing: http://www.elektronik.lodz.pl/gwinty/rystech/gwnp.jpg
For now it would be enough to select the existing hole on a drawing and choose "thread symbol" and it should appear on hole. Also additional dimensioning is needed for ISO and ANSI.

WandererFan, 2016-03-10:
will take me a while to digest all your comments, but thanks.  it's great
to get somebody else's view.

as for:
7) Unfortunately I cannot use cross section. Can you describe how to use
that?
8) I don't know how to use "hatched area"

7) this is currently only partially working.  it draws the section view,
but doesn't highlight the section face.  should be sorted by next week.
8) select a closed set of edges in a View, then click the Hatched Area
icon.

wf

WandererFan, 2016-03-16:
"3) I don't like that I need to choose horizontal or vertical dimensioning. Why? It should automatically decide is it vertical or horizontal or between corners."

If you pick the endpoints of an edge, how is the program going to know that you want to measure the vertical distance between endpoints, horizontal distance between endpoints, or the length of the edge itself?

WandererFan, 2016-03-16:
"9) Hidden lines should be dashed not light grey. I mean can be light grey but needs to be dashed ;)"

added #40.  There is already a preference for hidden line color.

WandererFan, 2016-03-16:
"1) What I liked is projection of model...It would be better to click first "projection button" then it will ask which model add to drawing."

This is about the ProjectionGroup task dialog?   There was a big debate in the forums about action-object vs object-action user interactions.  I thought the consensus was that FC was object-action (ie pick an object, then say what you want to do with it)).   I could be wrong.

WandererFan, 2016-03-16:
"5) You can move views, that's a large step but if you click on left side menu to "X", "Y", "tolerance" it is going back on place (destroys whole view sets)."

added #41.

WandererFan, 2016-03-16:
"2) Good is that you can choose scale while "projection". Not necessary good is that naming under view for example: "front", "bottom" etc.. good I can remove that. Instead that, it will be good when you are choosing different scale than "page scale", should text appear something like this for example "Scale 1:1", because if scale is set for page you can add to drawing other views with different scale and then under text is necessary."

Don't see how program can predict what the use wants to see in the Label property.  Labels are not displayed in finished Drawing anyway (see ToggleFrames command).

You can use Annotation to add notes - such as "Scale 1:1".

WandererFan, 2016-03-16:
"6) I don't like that you can go out with a view over the edge of "paper" but you cannot pan. Looks like card is centred on a screen and cannot move over grey area."

Page is centered.  You can pan L-R, Up-Down with the scroll bars after you Zoom in.

WandererFan, 2016-03-16:
"Next thing when you are rotating "main" view by custom angle (that's great) you should have option to project object by this angle..."

Not sure I understand this one.  Do you want the projection angle == 3D view angle when creating a ProjectionGroup?

efyx83, 2016-03-25:
I don't want to make mess and create new topic. Here are some drawing modules for another 3d CAD program. Take a look. Very nice is solved "broken out section" in Ironcad. Some links below:
https://www.youtube.com/watch?v=z5wp6qzH9yU (this is early module for Ironcad)
https://www.youtube.com/watch?v=m1JQZKNS3hU - evolved to this, BTW auxiliary view also is very welcome :)
https://www.youtube.com/watch?v=vGmj3EdGkP4 - broken out section
and Solidworks - the best drawing module I've worked with, see how dimensions are created, the views, details etc.. https://www.youtube.com/watch?v=PZeBjTeAZh8
Hopefully you will have some time to look and maybe you will use those ideas into your module

ianrrees, 2016-03-25:
Thanks for the links efyx83!

For me at least, it's much easier to deal with one topic per issue, so tacking this on the end is actually more messy.  Plus, we've got an item for broken out views - https://github.com/WandererFan/FreeCAD-TechDraw/issues/34

WandererFan, 2016-03-26:
Wow, lots of slick Gui stuff in the links.  Might have to wait for v2.0 ;)

WandererFan, 2016-06-16:
This is mostly fixed or broken out into other issues, so I'm closing it.

closed 2016-06-16 by WandererFan