WandererFan/FreeCAD-TechDraw #42: Links to 3D Geometry for True Dimensions

WandererFan, 2016-03-24:
The original 3D <-> drawing automatic link logic was not reliable.  We need a way to create references from Drawing Geometry to 3D geometry.  

Options: 
1) try to automatically link 3D edges/vertices to 2D counterparts.  This appears to be computationally intense. 
2) create new functions to manually link 3D/2D geometry.
3) link Drawing Dimension to Draft Dimension and/or "Measure" 
4) ????

ianrrees, 2016-03-24:
Hmm, seems like another place where topological naming is going to be useful.  I wonder if it makes sense for us to add attributes to the original 3D object, and preserve those attributes in the 2D representation?  If this were architected well, then perhaps it could use the topological naming system once that exists.

WandererFan, 2016-03-26:
Honestly, Topological Naming seems like the Philosopher's Stone to me.
It's going to fix everything.

I hadn't thought of making 3D point to 2D.  I guess because the original
was all 2D -> 3D.

ianrrees, 2016-03-27:
I wasn't thinking so much about the 3D pointing to 2D, rather if all the elements of the 3D object had some sort of persistent identifier (GUID or whatever), then we could somehow hang on to those identifiers along with the 2D elements.  Or, is that 2D->3D identifier what you're referring to as the link logic?

WandererFan, 2016-03-27:
Our only link now is PropertyLink to Source 3D object + integer 3D
Edge/Vertex number
for each piece of 2D geometry.  3D Edge number is sometimes null, sometimes
multiple 2D geoms point to single 3D geom, and this is at best
semi-persistent.

I've had a few ideas about creating the links, but so far they all seem to
be a lot of mouse clicks or unreliable.

WandererFan, 2016-04-19:
fixed, but see #46

closed 2016-04-19 by WandererFan