WandererFan/FreeCAD-TechDraw #44: C++11

ianrrees, 2016-03-28:
@WandererFan, what do you think about using a bit more C++11 in TechDraw?  I'd personally like to use it, but know you're still running an older gcc so it might be a pain to switch on your end.

WandererFan, 2016-03-28:
I'm just trying to recompile with g++-4.9 (+std=C++11) and boost 1.55.
I'll let you know how I make out.

WandererFan, 2016-03-28:
Compiled successfully with g++-4.9 and boost 1.55.  Had to make a change to
Inventor/SbBasic.h to get it to work.  Looks like this error:
http://forum.freecadweb.org/viewtopic.php?t=14108&start=10
 (cc_debugerror_post).  Think that is boost related, not c++11.  Also lots
of warnings in other modules.

So I can work with c++11.  Will it be a problem for testers?

ianrrees, 2016-03-28:
I don't imagine it'll be a problem for too many people - if it does cause problems, they're just foreshadowing issues with building the FreeCAD master soon.

We'll see how it goes, shouldn't be a big deal to back out problematic changes.

closed 2016-03-28 by ianrrees