WandererFan/FreeCAD-TechDraw #45: C calling conventions used for C++ objects.

ianrrees, 2016-04-03:
I think we should clean this up in a more C++ - style way, but probably won't have time to do it for at least a week or two.

freecad-code/src/Mod/TechDraw/App/Geometry.h:225:42: warning: 'chainGeoms' has
      C-linkage specified, but returns user-defined type 'std::vector<TechDrawGeometry::BaseGeom *>' which
      is incompatible with C [-Wreturn-type-c-linkage]

freecad-code/src/Mod/TechDraw/App/Geometry.h:226:18: warning: 'nextGeom' has
      C-linkage specified, but returns user-defined type 'TechDrawGeometry::getNextReturnVal' which is
      incompatible with C [-Wreturn-type-c-linkage]

WandererFan, 2016-04-14:
I don't get this warning with gcc-4.9.2 on Ubuntu 12.04.  

AFAIK I haven't set any extra/different flags in CMake.  Do you have extra warning options set?

ianrrees, 2016-04-14:
I think my system is more or less default in terms of the warning/error flags.  Probably just a clang/gcc difference - will take care of this sometime as I can easily tell when it goes away.

WandererFan, 2016-04-28:
App/DrawUtil also uses extern C

closed 2016-05-06 by ianrrees