WandererFan/FreeCAD-TechDraw #50: TechDraw compilation - no workbench

efyx83, 2016-05-09:
Hi,
I compiled two times source from https://github.com/WandererFan/FreeCAD-TechDraw and I couldn't find techdraw module. Any idea why. I configure it with cmake-gui (no option techdraw to select, all others modules are available).

ianrrees, 2016-05-09:
Sounds like you have the git repository's "master" branch checked out, but need the "TechDraw" one.  In a terminal, if you cd to the git repository (it'll likely be called some-path/FreeCAD-TechDraw ) and do "git status" it will tell you what branch you're on.  "git checkout TechDraw" will put you on the TechDraw branch.

There are also GUI programs for working with git, I use SourceTree a bit and like it.

WandererFan, 2016-05-13:
The TechDraw build option is at line 201 in the main CMakeLists.txt.

WandererFan, 2016-05-16:
We can reopen this one if it is still a problem, but I think there's nothing for us to do.

closed 2016-05-16 by WandererFan

liutas4x4, 2016-07-21:
Hi

After unzipping from FreeCAD_0.17.7745_x64_dev_win.7z I cannot fount TechDraw workbench too

WandererFan, 2016-07-21:
I've posted your issue on the FreeCAD forum: http://forum.freecadweb.org/viewtopic.php?f=4&t=16687
Check there for updates.

Thanks.