WandererFan/FreeCAD-TechDraw #51: Circles

joukeHijlkema, 2016-05-20:
Hi,

Thank you for attacking this. A good tech drawing module has been missing from this wonderful project.
I like where this is going. However I have 2 questions.

1. How do I get a horizontal (or vertical) dimension to attach to a circle center point (to indicate the position of a hole in a plate for instance)

2. How do I make a hatch pattern (what are the criteria ? I tried but to no avail)

WandererFan, 2016-05-20:
1) good point.  we don't plot the centre point, so you can't use a vertex to vertex measurement.  We haven't gotten around to datum lines & point yet either.  I'll add it to the enhancement list.
2) select a closed region of edges, then press the hatch button.  you may need to press recompute.  if you don't have a default pattern set you'll also have to change the pattern property.  There are svg hatch patterns in Mod/Draft/Resources/patterns in the source repository.  Let me know how you make out with this.  We have some issues with the Edges we get from OCC not always intersecting which breaks the code. 

Thanks for the feedback!

WandererFan, 2016-07-09:
#52

closed 2016-07-09 by WandererFan