WandererFan/FreeCAD-TechDraw #52: Drill Hole Location (Dimension)

WandererFan, 2016-05-20:
We need a way to create a dimension from the center of a circle to a point on the Part.   We don't plot the centre point, so you can't use a vertex to vertex measurement. We haven't gotten around to datum lines & points yet either so you can't use that.  

thanks to joukeHijlkema!

plot a vertex at circle centre if ShowCenterMarks is true?

ianrrees, 2016-05-20:
I like the idea of plotting the centre by default.  In my imagined usage of a Drawing, holes would be made with a drill press or mill or whatever, so would need to locate the centre point based on measurable dimensions on the the real object.

WandererFan, 2016-05-20:
There is logic to put a centre mark, but we need a vertex to do
dimensioning.

WandererFan, 2016-08-02:
Done.  Circle centres can now have marks and if the have a mark they are treated like a vertex and can be used in dimensions.   We also have a vertex-to-edge dimension now, so I think this one should be closed.

closed 2016-08-02 by WandererFan