WandererFan/FreeCAD-TechDraw #53: Crash when starting the "new section view" function

SebXcxl, 2016-05-31:
Hi,

I use the freshly compiled without errors Freecad v0.17 rev 7760 on Win7 64bit.
When I start the function "Insert a new section View of the part" of the techdraw module, Freecad crashes every time, for any object selected.
The .fcstd and a screenshot just before the crash :

[Freecad_crash.zip](https://github.com/WandererFan/FreeCAD-TechDraw/files/290442/Freecad_crash.zip)

To reproduce :
1) Start Freecad
2) File / New
3) Open the Part design module
4) In tasks ; "Create body"
5) Then "Create sketch"
6) "XY plane"
7) Draw a rectangle from origin with any dimension (something like 2 or 3 mm for each side)
8) Close sketch
9) Select it and "Pad" function on it (2 or 3mm)
10) Open the Techdraw module
11 ) "Insert new drawing page from template" and select the A3-landscape.svg of the drawing module
12) Select "Drawing" and "Pad" in the project tree
13) Click on "Insert a new section view"
14) Crash (on my computer)

WandererFan, 2016-05-31:
Thanks for the test case.  Will check it out.

WandererFan, 2016-05-31:
I wonder why your revision is +71 compared to me (7689 vs 7760)?

I don't get a crash.  I get an error on the SectionView (red icon in tree) because the default section plane doesn't cut the Pad.  If I move the section plane origin slightly along +Z, I get a section view. 

When I open your cadTest.fcstd file from FreeCAD_crash.zip, I get an error message from PartDesign(?) 'Can't find Origin for "Body"', but if I make a SectionView from Pad after that it behaves as above. 

Different OCC version? Build type? I wonder why your revision is +71 compared to me (7689 vs 7760)?

OS: Ubuntu 12.04.5 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.7689 +79 (Git)
Build type: Unknown
Branch: TechDraw
Hash: 7a098e51967ad581ddc20cff32f7f4c0c57db4f3
Python version: 2.7.3
Qt version: 4.8.1
Coin version: 3.1.3
OCC version: 6.8.0.oce-0.17-dev

SebXcxl, 2016-06-01:
Ok this different version can make the crash happen. I will download the
source again with the git console  (maybe a did somethg wrong with gitHub) :

git clone https://github.com/WandererFan/FreeCAD-TechDraw.git free-cad-code
cd free-cad-code
git checkout TechDraw

and compile it again.

SebXcxl, 2016-06-01:
[image: about.png]Ok downloaded and compiled again with :

git clone https://github.com/WandererFan/FreeCAD-TechDraw.git free-cad-code
cd free-cad-code
git checkout TechDraw

And the about window attached.

But still crashing when I click on "Insert a section view"
It there a log somewhere? I can build the Debug x64 version if needed.

S.L.

SebXcxl, 2016-06-01:
Ok so with the debug version of Freecad I can open the report view. But
with this version I have another crash when I try to insert a new drawing
page (like A3 landscape or A4 landscape).
I attached you the last report I could get, and a screenshot of the error.

 [image: error.png]

Run Gui init script
Init: Running FreeCADGuiInit.py start script...
Init:   Searching modules...
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Sketcher... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\ReverseEngineering... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Image... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\PartDesign... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Measure(InitGui.py not found)... ignore
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Ship... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Plot... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\MeshPart... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\OpenSCAD... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Fem... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Spreadsheet... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Start... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Points... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Draft... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Test... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Import... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\TechDraw... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Complete... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Web... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Material... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Raytracing... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Mesh... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Path... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Arch... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Robot... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Part... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Idf(InitGui.py not found)... ignore
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Inspection... done
Init:      Initializing C:\Users\SL\Desktop\myTechDraw2\build\Mod\Drawing... done
Init: Loading FreeCAD GUI
Init: Running FreeCADGuiInit.py start script... done
Init: Activating default workbench StartWorkbench
Loading GUI of Drawing module... done
Init: Showing main window
Main window restored
Show main window
Toolbars restored
3Dconnexion device not attached.
Init: Entering event loop
Init: Processing command line files
Active view is Unnamed : 1[*] (at 0000000006B66A80)
Active document is Unnamed (at 0000000002C97880)
Module: Part
Loading Part module... done
Loading Sketcher module... done
Loading GUI of Part module... done
Loading GUI of Sketcher module... done
Loading GUI of PartDesign module... done
Loading PartDesign module... done
Sel : Clear selection
Sel : Add Selection "Unnamed.Body.(null)(0.000000,0.000000,0.000000)"
make ordering: 
X_Axis dep on:[touched] => Recompute feature
Y_Axis dep on:[touched] => Recompute feature
Z_Axis dep on:[touched] => Recompute feature
XY_Plane dep on:[touched] => Recompute feature
XZ_Plane dep on:[touched] => Recompute feature
YZ_Plane dep on:[touched] => Recompute feature
BodyOrigin dep on:[touched] => Recompute feature
Body dep on:[touched] => Recompute feature
[... 111 lines cut ...]
Sel : Clear selection
Solv: Executing Feature: Pad
ViewProvider update time: 0.029000 s
Shape tria info: Faces:6 Edges:12 Nodes:32 Triangles:12 IdxVec:36
make ordering: 
X_Axis dep on:
Y_Axis dep on:
Z_Axis dep on:
XY_Plane dep on:
XZ_Plane dep on:
YZ_Plane dep on:
BodyOrigin dep on: X_Axis Y_Axis Z_Axis XY_Plane XZ_Plane YZ_Plane
Sketch dep on: XY_Plane
Pad dep on:[touched] => Recompute feature
Body dep on: BodyOrigin Sketch Pad[touched] Pad[touched] => Recompute feature
Have to recompute the following document objects
  Body
  Pad
Solv: Executing Feature: Pad
ViewProvider update time: 0.015000 s
Shape tria info: Faces:6 Edges:12 Nodes:32 Triangles:12 IdxVec:36
Solv: Executing Feature: Body
ViewProvider update time: 0.018000 s
Shape tria info: Faces:6 Edges:12 Nodes:32 Triangles:12 IdxVec:36
Loading GUI of TechDraw module... done

WandererFan, 2016-06-01:
If you can get us a backtrace, that would be great, since neither Ian nor I have access to a Windows environment.  Do you have a debugger (ex gdb, WinDbg, ...) installed?

SebXcxl, 2016-06-01:
Not sure what I am doing because that's the first time I use a debugger,
but I think I have the backtrace.
The code stop at :

template <class BidiIterator, class Allocator, class charT, class traits>
bool regex_search(BidiIterator first, BidiIterator last,
                  match_results<BidiIterator, Allocator>& m,
                  const basic_regex<charT, traits>& e,
                  match_flag_type flags = match_default)
{
   return regex_search(first, last, m, e, flags, first);
}

template <class BidiIterator, class Allocator, class charT, class traits>
bool regex_search(BidiIterator first, BidiIterator last,
                  match_results<BidiIterator, Allocator>& m,
                  const basic_regex<charT, traits>& e,
                  match_flag_type flags,
                  BidiIterator base)
{
   if(e.flags() & regex_constants::failbit)
      return false;

   re_detail::perl_matcher<BidiIterator, Allocator, traits> matcher(first,
last, m, e, flags, base);
   return matcher.find();
}

In the file regex_search.hpp. I attached the log.

S.L.

> If you can get us a backtrace, that would be great, since neither Ian nor
> I have access to a Windows environment. Do you have a debugger (ex gdb,
> WinDbg, ...) installed?

 	msvcr120d.dll!000007fedd2fe190()	Unknown
 	msvcr120d.dll!000007fedd2ecddd()	Unknown
 	msvcr120d.dll!000007fedd2ed002()	Unknown
 	msvcr120d.dll!000007fedd2ecfb3()	Unknown
 	boost_regex-vc120-mt-gd-1_55.dll!000007fedd3f4d1e()	Unknown
 	boost_regex-vc120-mt-gd-1_55.dll!000007fedd4566dc()	Unknown
 	boost_regex-vc120-mt-gd-1_55.dll!000007fedd45654a()	Unknown
 	boost_regex-vc120-mt-gd-1_55.dll!000007fedd455a23()	Unknown
>	TechDraw_d.pyd!boost::regex_search<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > >,std::allocator<boost::sub_match<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > > >,char,boost::regex_traits<char,boost::w32_regex_traits<char> > >(std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > first, std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > last, boost::match_results<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > >,std::allocator<boost::sub_match<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > > > > & m, const boost::basic_regex<char,boost::regex_traits<char,boost::w32_regex_traits<char> > > & e, boost::regex_constants::_match_flags flags, std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > base) Line 56	C++
 	TechDraw_d.pyd!boost::regex_search<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > >,std::allocator<boost::sub_match<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > > >,char,boost::regex_traits<char,boost::w32_regex_traits<char> > >(std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > first, std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > last, boost::match_results<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > >,std::allocator<boost::sub_match<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > > > > & m, const boost::basic_regex<char,boost::regex_traits<char,boost::w32_regex_traits<char> > > & e, boost::regex_constants::_match_flags flags) Line 42	C++
 	TechDraw_d.pyd!TechDraw::DrawSVGTemplate::execute() Line 210	C++
 	TechDraw_d.pyd!TechDraw::DrawSVGTemplate::onChanged(const App::Property * prop) Line 133	C++
 	FreeCADApp_d.dll!App::Property::hasSetValue() Line 117	C++
 	FreeCADApp_d.dll!App::PropertyMap::setValues(const std::map<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,std::less<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,std::basic_string<char,std::char_traits<char>,std::allocator<char> > > > > & map) Line 1690	C++
 	TechDraw_d.pyd!TechDraw::DrawSVGTemplate::onChanged(const App::Property * prop) Line 120	C++
 	FreeCADApp_d.dll!App::Property::hasSetValue() Line 117	C++
 	FreeCADApp_d.dll!App::PropertyString::setValue(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & sString) Line 1289	C++
 	FreeCADApp_d.dll!App::PropertyString::setPyObject(_object * value) Line 1321	C++
 	TechDraw_d.pyd!TechDraw::DrawSVGTemplatePy::setCustomAttributes(const char * attr, _object * obj) Line 37	C++
 	TechDraw_d.pyd!TechDraw::DrawSVGTemplatePy::_setattr(char * attr, _object * value) Line 359	C++
 	FreeCADBase_d.dll!Base::PyObjectBase::__setattr(_object * obj, char * attr, _object * value) Line 167	C++
 	python27_d.dll!PyObject_SetAttr(_object * v, _object * name, _object * value) Line 1252	C
 	python27_d.dll!PyEval_EvalFrameEx(_frame * f, int throwflag) Line 2018	C
 	python27_d.dll!PyEval_EvalCodeEx(PyCodeObject * co, _object * globals, _object * locals, _object * * args, int argcount, _object * * kws, int kwcount, _object * * defs, int defcount, _object * closure) Line 3265	C
 	python27_d.dll!PyEval_EvalCode(PyCodeObject * co, _object * globals, _object * locals) Line 679	C
 	python27_d.dll!run_mod(_mod * mod, const char * filename, _object * globals, _object * locals, PyCompilerFlags * flags, _arena * arena) Line 1377	C
[... cut ...]
 	QtGuid4.dll!QGuiEventDispatcherWin32::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 1212	C++
 	QtCored4.dll!QEventLoop::processEvents(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 150	C++
 	QtCored4.dll!QEventLoop::exec(QFlags<enum QEventLoop::ProcessEventsFlag> flags) Line 204	C++
 	QtCored4.dll!QCoreApplication::exec() Line 1225	C++
 	QtGuid4.dll!QApplication::exec() Line 3824	C++
 	FreeCADGui_d.dll!Gui::Application::runApplication() Line 1816	C++
 	FreeCAD_d.exe!main(int argc, char * * argv) Line 236	C++
 	[External Code]

SebXcxl, 2016-06-02:
And the error when I start the Freecad release and I click on "Insert a
section view" :

[image: freecadError.png]

S.L.

WandererFan, 2016-06-02:
Something wrong with upload of "freecadError.png"?

en tout cas,  this is Confirmed.  There is a problem with the parsing of the EditableTexts for the A3Landscape.svg template . 

You can get an A3 template that works from github/Wandererfan/FCDraw2Test/Template/A3_LandscapeISO7200.svg.  You could also use the A3_LandScapeISO7200.svg template from the regular distribution, but the Edit handles will be in the wrong place. 

Thanks for your help in tracking this down. 

wf

ianrrees, 2016-06-02:
Hmm, IIRC the A3Landscape.svg is the template I had modified for the previous version of editable template texts (to add the dimensions for the edit box).  Wonder if that's related?

SebXcxl, 2016-06-03:
Indeed, with this SVG template FreeCAD_d.exe is not crashing when loading
the new page but some errors saying "Invalid SVG template syntax" when I
add a view (not section view) :

*Cannot read file
'C:/Users/SL/AppData/Local/Temp/FreeCAD_Doc_66ad0366-90a0-4f4e-a0ba-1e787000b472_eddb00_6504/A3_Landscape_ISO7200.svg',
because: (line 6)*

And still crashing for inserting a new section view :

*First-chance exception at 0x0000000076D1793B (ntdll.dll) in FreeCAD.exe:
0xC0000005: Access violation writing location 0x00000000000F0EE8.Unhandled
exception at 0x0000000076D1793B (ntdll.dll) in FreeCAD.exe: 0xC0000005:
Access violation writing location 0x00000000000F0EE8.*
In the debugger.

SebXcxl, 2016-06-03:
Ok, I don't know why but since a install Visual Studio dll's debug symbols,
FreeCAD_d.exe is not crashing anymore ; this is I got in the report window
when I click on section view (got a blank page drawing with a small dot
rectangle on it) :

*DrawViewSection::execute - building Section shape failed: Bnd_Box is void*

*DrawViewSection::execute - building Section shape failed: Bnd_Box is void*

*DrawViewSection::execute - building Section shape failed: Bnd_Box is void*

*make ordering: *

*X_Axis dep on:*

*Y_Axis dep on:*

*Z_Axis dep on:*

*XY_Plane dep on:*

*XZ_Plane dep on:*

*YZ_Plane dep on:*

*BodyOrigin dep on: X_Axis Y_Axis Z_Axis XY_Plane XZ_Plane YZ_Plane*

*Sketch dep on: XY_Plane*

*Pad dep on: Sketch*

*Body dep on: BodyOrigin Sketch Pad Pad*

*Template dep on:[touched] => Recompute feature*

*Section dep on:[touched] => Recompute feature*

*Page dep on:[touched] => Recompute feature*

*Have to recompute the following document objects*

* Section*

* Page*

* Template*

*Solv: Executing Feature: Template*

*Invalid SVG syntax in Template - Check EditableTexts*

*Solv: Executing Feature: Section*

*DrawViewSection::execute - building Section shape failed: Bnd_Box is void*

*Bnd_Box is void*

*Solv: Executing Feature: Page*

*TODO: Unimplemented function DrawPage::onChanged(Template)*

*Cannot read file
'C:/Users/SL/AppData/Local/Temp/FreeCAD_Doc_66ad0366-90a0-4f4e-a0ba-1e787000b472_eddb00_6600/A3_Landscape_ISO72001.svg',
because: (line 6)*

WandererFan, 2016-06-03:
"DrawViewSection::execute - building Section shape failed: Bnd_Box is void*

*Bnd_Box is void*

This part is ok.  It means the default section plane doesn't cut the Part.
Moving the section plane should fix it.

I'll try to look at the Templates this weekend.  There are some changes
between Drawing and TechDraw.  Maybe I didn't catch them all.

WandererFan, 2016-06-05:
Templates that work with TechDraw are now in a TechDraw/Templates directory.  Their names are *TD.svg.  I made an A3_LandscapeTD.svg that seems to work ok. 

Some of the Drawing templates cause problems with boost::regex in TechDraw.  I don't know why they work in Drawing but not in TechDraw. 

Section view now uses the middle of the Source's bounding box if the SectionOrigin is outside the box, so *Bnd_Box is void* should be solved too.

ianrrees, 2016-06-05:
With regards to the regex in TechDraw, I wonder if the change in FeatureSVTTemplate.cpp from this commit might be related: https://github.com/ianrrees/FreeCAD_tinkering/commit/6bf815bf8d90a23b641fda058407ce8b56907d38 - I have vague memories of problems with those regular expressions, I think around places where there are constructs like [^abc]*...[^xyz]*, but can't remember for sure.

Hoping to get to work on TechDraw this evening - got some chores that need to be done first though.

SebXcxl, 2016-06-05:
> ianrrees/FreeCAD_tinkering@6bf815b

Hi,

Thanks for the fixes, I will start a rebuild at my work tomorrow morning (I
don't have so much pressure these days^^) and give you à feedback.

SL

WandererFan, 2016-06-05:
regex was actually failing in execute(), not getEditableTextFromTemplate(),
so it probably isn't that commit.

The weirdest thing is that the TD code in execute looks identical to the
Drawing code, but ours doesn't always work.

I guess if we were clever, we wouldn't have 3 pieces of code to
find/replace the editable fields or 2 in the same compiland! if any of my
profs were still alive they'd be deducting points for sure.

SebXcxl, 2016-06-06:
Ok the SVG templates works and it is possible to add a section view to the
drawing. I couldn't be able to correctly modify the section plane to get
the section I wanted, but it displayed a blue rectangle with some empty
space around, so I suppose it's working ; will try again later. Thanks a
lot ;)

Sebastien L.

WandererFan, 2016-06-06:
de rien.   different SectionOrigins should work out like this:

​
I'm going to close this issue as it is getting quite long.  Please open
another with any other problems you find.

merci,
wf

​

WandererFan, 2016-06-06:
There's supposed to be a picture in the post above.  Differences between gmail & github I guess.

closed 2016-06-06 by WandererFan