WandererFan/FreeCAD-TechDraw #55: Unicode trouble in text fields

mumme74, 2016-07-02:
I tried TechDraw today, thank you for working on this.
I did find some issues with swedish(non english) chars in Title-, Description fields etc.
such as: åäöÅÄÖ

I tried this fix which seems to work: 

```
diff --git a/src/Mod/TechDraw/Gui/DlgTemplateField.cpp b/src/Mod/TechDraw/Gui/DlgTemplateField.cpp
index 4e046d0..eb97cd4 100644
--- a/src/Mod/TechDraw/Gui/DlgTemplateField.cpp
+++ b/src/Mod/TechDraw/Gui/DlgTemplateField.cpp
@@ -49,20 +49,20 @@ void DlgTemplateField::changeEvent(QEvent *e)
 
 void DlgTemplateField::setFieldName(std::string name)
 {
-    QString qs = QString::fromStdString(name);
+    QString qs = QString::fromUtf8(name.data(), name.size());
     lblName->setText(qs);
 }
 
 void DlgTemplateField::setFieldContent(std::string content)
 {
-    QString qs = QString::fromStdString(content);
+    QString qs = QString::fromUtf8(content.data(), content.size());
     leInput->setText(qs);
 }
 
 std::string DlgTemplateField::getFieldContent()
 {
     QString result = leInput->text();
-    return result.toUtf8().constData();
+    return result.toStdString();
 }
 
 #include "moc_DlgTemplateField.cpp"
```

Hope it helps...

WandererFan, 2016-07-02:
Fix works great.  Thanks.

WandererFan, 2016-07-04:
need to revisit for non-latin1 chars.   Qt::escape uses latin1 internally so we don't get the rest of the utf8 chars. Qt5 has a fix.

closed 2016-07-04 by WandererFan