WandererFan/FreeCAD-TechDraw #60: Preselection & Selection Colors

ickby, 2016-07-13:
Preselection color and selection color do not match the preference settings. Preselection in Black (and hence not visible when a black edge is preselected) and selection color is blue, while TechDraw preferences say yellow and green. 

IMHO it woul dmake sense to not have a custom color preference but to use the generall frecad preselection and selection color user setting also for tech draw.

WandererFan, 2016-07-13:
I can't reproduce this here - I get black/green/yellow and I can change with preferences.  Coulld you please send me this section of your usr.cfg:

        <FCParamGroup Name="Mod">
          <FCParamGroup Name="TechDraw">
            <FCParamGroup Name="Colors">
              <FCUInt Name="NormalColor" Value="421075455"/>
              <FCUInt Name="SelectColor" Value="481107199"/>
              <FCUInt Name="PreSelectColor" Value="4278190335"/>  

The reason for a separate colour choice is that some people use odd backgrounds in the 3D view and they might need a different set against the white of the drawing page background.

WandererFan, 2016-07-13:
Typo in default preselect if no pref set:  0x00080800  sb 0x00808000  explains black preselect,
QGIPrimPath.cpp and QGIView.cpp

WandererFan, 2016-07-13:
- PrimPath/QGIView
  - default select is missing setpackedvalue
  - change of preferences is not reflected in objects previously created
    - works in newly created objects or after document close/reopen

ickby, 2016-07-13:
Ah yes, if a view is recomputed it takes the new color values. This seems very unintuitive. There is a callback when preferences change, I think in the ViewProvider,, maybe looking out for the selection colors there is a useful thing

WandererFan, 2016-07-15:
fixed in PR "Color Preferences #218".  Still need to wait for object repaint to see result.  

Need a "repaint" button??

closed 2016-07-15 by WandererFan